NWFSC-OA-lab/field_sensors #12: Web app

pmcelhany, 2022-04-26:
- [ ] Preliminary design of web app interface (2022.04.26)
- [ ] Review preliminary design

pmcelhany, 2022-04-26:
Update 
[Interface doc](https://docs.google.com/document/d/1fEuwa57X2if4j0FIOUwNqlYIXv_eCr1Zn2rY9eV4Enk/edit)

sjhwa, 2022-05-03:
- [ ] Get web server up and running and feed it sample data from Android app (5/4/2022)
- [ ] Implement Web Page for downloading CSV file for time-series data (5/72022)
- [ ] Provide Core UI features (Email Alerts, Selection of Data over periods of time in table-format) (5/10/2022)
- [ ] Server Cost analysis (5/10/2022) 
- [ ] Implement Additional UI Features (Sensor Selection, Data Visualization) (5/15/2022 <- original due date)

pmcelhany, 2022-05-03:
looks like a good list

pmcelhany, 2022-05-04:
alarms not essential until we get continuous live communication

pmcelhany, 2022-05-17:
2022.05.17

- connect with sebas

sjhwa, 2022-05-17:
- [ ] Set up server for public Traffic (5/17/2022)
- [ ] Send Sebastian HTTP code info (5/17/2022)
- [ ] Continue with UI Dev (and keep team updated through email!)

pmcelhany, 2022-05-24:
2022.05.24

- [ ] get url to sebastian (need to modify security) do this afternoon
- [ ] options for sensor calibration data, alarms, new measurements, lat-lon google map

sjhwa, 2022-05-26:
- [ ] Feed sample data from Android app for demo 
- [ ] Server side: fix missing parameters error
- [ ] Client side: add more chart options, column names for output file
- [ ] Client side: fix user log in screen (i.e. fix deprecated node modules)
- [ ] Feed sample data from Android app for demo

Documentation:
- [ ] analyze server costs, domain name, and possible database needs
- [ ] 
- [ ] 
- [ ] Continue server Handle

pmcelhany, 2023-01-10:
Capstone 22 issue

closed 2023-01-10 by pmcelhany