NWFSC-OA-lab/field_sensors #8: Bench test durafet with labview

pmcelhany, 2022-04-26:
- [ ] Wire plug-in power supply,, durafet, labjack, computer
- [ ] Test durafet with windows and labjack software
- [ ]  Read labjack with aurdino

pmcelhany, 2022-04-26:
Will plan to start step one on 2022.04.26 at 4:00
@michaelmaher114 Can you join the UW team to talk about durafet and labjack, etc. at 4:00 today? perhaps in person at UW oceanograhy?

pmcelhany, 2022-04-26:
Research purchase of 24 bit A-to-D aurdino module.

pmcelhany, 2022-04-26:
@NWFSC-OA-lab/field-sensors-team Mike can join your durfet wiring meeting at 4:00 today. I sent a calendar invite.

AaronDyer13, 2022-04-26:
> @NWFSC-OA-lab/field-sensors-team Mike can join your durfet wiring meeting at 4:00 today. I sent a calendar invite.

Sounds good, thanks!

pmcelhany, 2022-05-04:
- Currently testing with 16bit atd (weird results?)
- need to get 24bit (get it working with 16bit if 24 bit not available). Should be easy to swap atod modules.
- 16bit a to d are pretty available

pmcelhany, 2022-05-10:
2022.05.10 

- reading voltage into

pmcelhany, 2022-05-10:
@michaelmaher114 Do we have some tris for durafet calibration we can use for the sensor project?

pmcelhany, 2022-05-17:
2022.05.17
same tasks as last week

pmcelhany, 2023-01-10:
Capstone 22 issue

closed 2023-01-10 by pmcelhany