NWFSC-OA-lab/field_sensors #21: Hardware management

pmcelhany, 2023-01-10:
- [ ] Inventory initial hardware
- [ ] Identify where hardware stored
- [ ] Identify how to transfer hardware

pmcelhany, 2023-01-17:
I have another box of hardware that I will bring to the meeting today. It include some Arduino boards and other components.

ajbuckingham00, 2023-01-17:
All hardware is currently stored at our desk in the capstone work space, which is under keycard access.

axiongH, 2023-01-18:
I'm currently working on creating an inventory list of all of our supplies, but I'll make a separate inventory list for the prototype's hardware

axiongH, 2023-01-21:
Added a google doc in our drive that keeps inventory of what we currently have in the boxes, as well as what the prototype is using