NWFSC-OA-lab/field_sensors #24: Version 1 electronics test

pmcelhany, 2023-01-10:
- [ ] Review schematic of prototype 1
- [ ] Verify components/wiring of prototype 1 board
- [ ] Work with Arduino code to test components on prototype 1 board

axiongH, 2023-01-21:
1/21: Tested most of the components individually. ADC1115 and Bluetooth Module are still under testing.
- ADC1115 requires a pull-up resistor for the SCL and SDA line to use the I2C
- Bluetooth module requires the use of an app so that we need to figure out how to use so this one is taking a bit longer

ajbuckingham00, 2023-01-24:
In addition to the items above, we fixed several soldering issues on the board. We were unable to fix the soldering on the ADC board (I think either the pads are lifted or they are corroded/damaged somehow, solder would not stick to them at all even after cleaning with alcohol), so we installed the 12 bit ADC instead and were able to use that.

axiongH, 2023-02-08:
We were able to connect to the app, however, we are not yet able to pull data from the SD card