NWFSC-OA-lab/field_sensors #27: Cellular communication

pmcelhany, 2023-01-10:
- [ ] Research cellular communication options

sakura-gen, 2023-01-24:
Researched the SIM7000 shield documentation. Looking to read more on it to see if it's a viable option. Also played around with mobile dev edge kit to see if we can integrate and use as a temporary stepping stone in enabling cellular comms.

Didn't find too much information on why last year's team didn't implement cellular. Found some info saying that it was because of SIM cards. Looking for more info on that.

TO DO: Figure out barriers to implementing SIM7000 board. Look into documentation to see if we could send some test data in the next week. May need a Verizon or ATT SIM card to test (prepaid)

CraigNorrie, 2023-01-24:
Potential provider for internet access that last years team researched a little.

https://www.hologram.io/

ajbuckingham00, 2023-03-01:
Gen did some work with the sim7000 board, and was able to make some progress but identified that the adafruit fona board will be much easier to work with and most likely worth the $80 purchase