NWFSC-OA-lab/field_sensors #28: Housing review

pmcelhany, 2023-01-11:
- [ ] Review current housing
- [ ] Look a SAFS shop options
- [ ] Look 3-D printer

CraigNorrie, 2023-01-15:
We have just been offered up 6 aluminium cans that are rated for ~1000m depth ("I have 2 of the smaller ones and 5 of the bigger ones. They have O-rings and 2-3 holes each for terminator ends."). These may be a good option to adapt for this project. 

I will pick these up on Wednesday so once we have decided who will be working on housings we can organise a hand off of these. 

Also, on the housing end I have used t[his company, Blue robotics](https://bluerobotics.com/), to purchase cable pass throughs to get cables into the housing. 

[image]

[image]

ajbuckingham00, 2023-01-24:
https://docs.google.com/document/d/1tTLNvvjGGg6psoWKbbJDTTxCdVLFyZhsqk0whiyrYP8/edit

Housing research doc from last week, with several useful papers linked